suoapvs/NTC_Thermistor #1: Compilation failure

rdehuyss, 2018-05-25:
Hi Yurii,

I don't understand why (probably it is just too late ;-) ) but I'm having compilation problems:

Any idea what could be wrong?  The code is really readable by the way!

	NTC_Thermistor.cpp:13: error: prototype for 'NTC_Thermistor::NTC_Thermistor(int, double, void (*)())' does not match any in class 'NTC_Thermistor'

	 NTC_Thermistor::NTC_Thermistor(

	 ^

	In file included from c:\Projects\Arduino\Showerloop\NTC_Thermistor.cpp:11:0:

	NTC_Thermistor.h:60: error: candidates are: constexpr NTC_Thermistor::NTC_Thermistor(NTC_Thermistor&&)

	 class NTC_Thermistor final {

		   ^

	NTC_Thermistor.h:60: error: constexpr NTC_Thermistor::NTC_Thermistor(const NTC_Thermistor&)

	NTC_Thermistor.h:122: error: NTC_Thermistor::NTC_Thermistor(int, double, double, double, double, int, long int)

	   NTC_Thermistor(

	   ^

	NTC_Thermistor.h:101: error: NTC_Thermistor::NTC_Thermistor(int, double, double, double, double)

	   NTC_Thermistor(

	   ^

	showerloop:17: error: expected unqualified-id before 'const'

	 {

	  ^

	exit status 1

YuriiSalimov, 2018-05-28:
Hi rdehuyss,
Try using one of the following constructors:
```cpp
NTC_Thermistor(
  int pin,
  double referenceResistance,
  double nominalResistance,
  double nominalTemperatureCelsius,
  double bValue
);
```
or 
```cpp
NTC_Thermistor(
  int pin,
  double referenceResistance,
  double nominalResistance,
  double nominalTemperatureCelsius,
  double bValue,
  const int readingsNumber,
  const long delayTime
);
```

rdehuyss, 2018-05-29:
My mistake - problem was with code completely unrelated and the Cpp compiler just put me on the wrong track.

closed 2018-05-29 by rdehuyss