suoapvs/NTC_Thermistor #6: How to get nominal resistance and nominal temperature

zsocietyGithub, 2019-03-12:
i just buy cheap termistortor with spec:  NTC 10k±1% 3950, 
and i dont know how to get nominal resistance and nominal temperature from this termistor.
anyone can help me or give me a reference,

thanks for advice,

YuriiSalimov, 2019-03-12:
Hello!

For NTC 10k±1% 3950:
nominalResistance = 10 kOhm
nominalTemperature = 25 С
bValue = 3950

Regards, Yurii.

zsocietyGithub, 2019-03-13:
> Hello!
> 
> For NTC 10k±1% 3950:
> nominalResistance = 10 kOhm
> nominalTemperature = 25 С
> bValue = 3950
> 
> Regards, Yurii.

Thanks for quick respons Yurii,

Yesterday I tried with the parameters as above as you gave. but the temperature reading does not match with the actual temperature (I compare it with a thermometer).
reading from a thermistor 25.50 ^ C and from a calibrated thermometer 27.00 ^ C.

my question is how are the procedures for calibration this termistor?
can I increase or decrease the value of the results so the result match with my termometer or is there another method?

thanks for advice

YuriiSalimov, 2019-03-13:
Hello!

You can try to slightly change the value of the input parameters.
Also, I made some changes to the code. Сheck for updates, and try again.

Regards, Yurii.

zsocietyGithub, 2019-03-14:
i will check the update...and try...

thanks

closed 2019-03-19 by YuriiSalimov