suoapvs/NTC_Thermistor #8: reupload 2.0.0 for platformio library

lackdaz, 2019-08-01:
I'm getting a compilation error regarding `timeoutInMillis` not being in scope for `AverageThermistor.cpp` when compiling in platformio. I checked your source files here and they seem to have been replaced with `delayTimeInMillis`. Perhaps reflect these changes on the platformio library as well?

YuriiSalimov, 2019-08-02:
Hi lackdaz.

I found your problem. This is a bug from the previous version. For unknown reasons, platformio has not uploaded the changes in version 2.0.0. Try to update the library to version 2.0.1.

Regards, Yurii.

closed 2019-08-05 by YuriiSalimov

lackdaz, 2019-08-06:
cheers mate, I fired up a test project and it looks like it compiled this time. I'll let you know for sure when I've removed the cloned library on my working project and start using the internal 2.0.1.

This is by the way, the most accurate and responsive thermistor library I've used

YuriiSalimov, 2019-08-07:
Thank!
Good luck with your programming!