suoapvs/NTC_Thermistor #17: wifi going down

sova32, 2024-04-29:
you use 
delay();
and
sleep();
func`s !
Seriously?
No one does that!
Have you heard about the millis() function?

suoapvs, 2024-04-29:
This code was written 5 years ago. I have no idea what you're talking about anymore. I'm no longer interested in changing anything. You can download the code and make your changes.

sova32, 2024-04-29:
Some lib`ss like AsyncWebServer or if you using wi-fi on your microcontrollers going down when delay() or sleep() present in code.
it is very goon lib but in
AverageThermistor.cpp
and
SmoothThermistor.cpp
you must use millies()
some like this
```
unsignet int delay = 1000; // 1 second for example
unsignet int preM = 0;
loop(){
    if (millies() >= preM+delay){
        // your code here
        preM = millies;
}
```
good luck bro!

sova32, 2024-04-29:
> This code was written 5 years ago. I have no idea what you're talking about anymore. I'm no longer interested in changing anything. You can download the code and make your changes.
Друже! ти з України?! 
Як ся маєш?
Я хотів би розпитати в тебе дещо про GoIT. Ми б не могли перемовитись парой слів?

suoapvs, 2024-04-29:
I can change sleep() function:

```
inline void AverageThermistor::sleep() {
  unsigned long start = millis();
  while (millis() - start < this->delayTime) {
    // wait
  }
}
```
Review the code, please

closed 2024-07-09 by suoapvs