suoapvs/NTC_Thermistor #19: Does not compile for non ESP32 platforms

joegatling, 2024-07-07:
I am running into an issue compiling version 2.1.0 for an Adafruit Trinket M0, which has a ATSAMD21E18 processor. When I try to compile it I get the following error:

`
.pio/libdeps/adafruit_trinket_m0/NTC_Thermistor/src/NTC_Thermistor.cpp: In member function 'virtual double NTC_Thermistor_ESP32::readVoltage()':
.pio/libdeps/adafruit_trinket_m0/NTC_Thermistor/src/NTC_Thermistor.cpp:125:17: error: 'analogReadMilliVolts' was not declared in this scope
  125 |  return (double)analogReadMilliVolts(this->pin) / (double)this->vref_mv * this->adcResolution;
      |                 ^~~~~~~~~~~~~~~~~~~~
`

I believe this is because the function analogReadMilliVolts() is an ESP32-sepcific function.

joegatling, 2024-07-07:
I have a PR with a fix: https://github.com/suoapvs/NTC_Thermistor/pull/20

suoapvs, 2024-07-09:
Done

closed 2024-07-09 by suoapvs